FirmWire/FirmWire #1: The MTK DB parser can not support the NR5G version

fenghao119, 2022-03-16:
I dumped the DB (dimensity-900) and decoded it with your code. It can be decoded as some segments, but no further decoding.

N3vv, 2022-03-16:
I have the same problem with dimensity-800.It looks like the format of these files(such as md1rom and md1_mddb) has changed a bit on the new soc.

mariusmue, 2022-03-16:
Thanks for your interested in FirmWire and the associated tooling.

Note that 5G modems are not officially supported by the framework; We started developing the framework before 5G modems were around.

The full list of tested/supported hardware models can be found here: https://firmwire.github.io/docs/quick_start.html
On MTK side, this would MT6762 (Helio P22) and MT6768 (Helio P65).

In any case, from what I recall, for 5G modems, MTK started to deploy MIPS I7200 cores. Instead of MIPS16e2, these cores would deploy nanoMIPS, so heavy changes in the emulation engine may be required to support emulation of those. (Note, my memory may be wrong here, it was a while ago when I looked into this.)

Either way, we would be happy to accept PullRequests extending the current functionality of the framework.

grant-h, 2022-03-18:
The database I believe changed to RocksDB format. We had some code that was for prototyping. We may bring this to production in the future. Can you provide the exact firmware images you were playing with so we can test later on?

N3vv, 2022-03-22:
Thanks a lot! Here is a  dimensity-800(MT6873) baseband firmware which is downloaded from a third-party website. In my previous analysis of its md1rom, I was puzzled by many GFH structures with the magic number "4D4D4D".
[MT6873](https://github.com/N3vv/BasebandFirmware/blob/main/MTK/dimensity/cannong_global_images_V12.0.6.0.QJEMIXM_20210125.0000.00_10.0_global.zip)