FirmWire/FirmWire #2: Field information of struct `qitem_lte_rrc`

MustBastani, 2022-03-17:
I am reading the [fuzz task for LTE RRC](https://github.com/FirmWire/FirmWire/blob/main/modkit/shannon/fuzzers/lte_rrc.c#L26), and I have some questions about the queue item structure used in this fuzzer.
1. You used `0xc3a0` for the field `op`. Is this the message ID that providing malformed `RRCConnectionReconfiguration-r8-IEs` causes crash?
2. What are the possible valid values for the field `pdu_type`?
3. Does the field `asn_pl` follows the following ASN1 format from `RRCConnectionReconfiguration` payload? [source](http://www.3gpp.org/DynaReport/36331.htm)
```
RRCConnectionReconfiguration ::= SEQUENCE {
rrc-TransactionIdentifier RRC-TransactionIdentifier,
criticalExtensions CHOICE {
c1 CHOICE{
rrcConnectionReconfiguration-r8 RRCConnectionReconfiguration-r8-IEs,
spare7 NULL,
spare6 NULL, spare5 NULL, spare4 NULL,
spare3 NULL, spare2 NULL, spare1 NULL
},
criticalExtensionsFuture SEQUENCE {}
}
}
```

grant-h, 2022-03-18:
1. These PDU types are implementation defined and are specific to the ASN.1 decoder used by Shannon. You can use GLINK with the --console to try sending some messages with different PDU types to see that different decoders (e.g. BCCH, DCCH) are chosen. I don't have a mapping handy.
2. `asn_pl` is a ASN.1 UPER encoded payload following the rules of 36.331. Depending on the PDU type is how it will be interpreted.

MustBastani, 2022-03-18:
Thank you for your comment. 
1. I also tried GLINK with `send_rrc`, but I don't know what are the input formats (`payload` and `opcode`). I tried the following cases in separate sessions:
```python
gl = self.get_peripheral('glink')
asn_pl = b"\x20\x1b\x3f\x80\x00\x00\x00\x01\xa9\x08\x80\x00\x00\x29\x00\x97\x80\x00\x00\x00\x01\x04\x22\x14\x00\xf8\x02\x0a\xc0\x60\x00\xa0\x0c\x80\x42\x02\x9f\x43\x07\xda\xbc\xf8\x4b\x32\x18\x34\xc0\x00\x2d\x68\x08\x5e\x18\x00\x16\x80\x00"

# 1
gl.send_rrc(asn_pl, 0)

# 2
gl.send_rrc(asn_pl, 0xc3a0)

# 3
op = 0xc3a0
qitem_size = 0x10
unused = 0
pdu_type = 0
gl.send_rrc(struct.pack("<IIIII", op, qitem_size, unused, pdu_type, len(asn_pl)) + asn_pl, 0xc3a0)
```

All of the cases resulted in the following lines from `LteRrc` task:
```powershell
[LteRrc] 0x414dd023 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]DrxStart: gDrxRrc_Flag 0 gDrxL1_Flag 1 gDrxRrc_SaveL1Flag 1
[LteRrc] 0x40cc09b3 0b1: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_CommDb.c] - [MAIN]LTERRC_WRONG_ARGUMENT(ConvertMsgId:933237)
[LteRrc] 0x414dbabd 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][ProcessMsg] Fored Set -> main to Main
[LteRrc] 0x40d96bdd 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_SuppSvcFrameworkCore.c] - [MAIN]_SuppSvcSetCurrentActive To NULL
[LteRrc] 0x414dbb19 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][ProcessMsg] Fored Set -> main to cur active
[LteRrc] 0x40d95695 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_SuppSvcFrameworkCore.c] - [MAIN]_SuppSvcNo Active Scv Force to LTERRC_SUPP_SVC_MAIN
[LteRrc] 0x414e7dab 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_ProcDsds.c] - [MAIN][LTE RRC DSRC] LteRrcDsds_CheckIsProcStart msgtype(10)
[LteRrc] 0x414e8039 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_ProcDsds.c] - [MAIN]LteRrcDsds_CheckIsProcStart :: Invalid Message Type[10]
[LteRrc] 0x40d95695 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_SuppSvcFrameworkCore.c] - [MAIN]_SuppSvcNo Active Scv Force to LTERRC_SUPP_SVC_MAIN
[LteRrc] 0x414dd32f 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]SET OPFLAG :(MAIN-SUPP_SVC_CANCLE_DISPATCH)
[LteRrc] 0x40d95695 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_SuppSvcFrameworkCore.c] - [MAIN]_SuppSvcNo Active Scv Force to LTERRC_SUPP_SVC_MAIN
[LteRrc] 0x414d4b71 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Free LinkMsg : 0xff000000
[LteRrc] 0x414d4bc5 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Free Rx Msg
[LteRrc] 0x414dd7e9 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]CLEAR OPFLAG :(MAIN-STATE_CHANGE_TRIGGER)
[LteRrc] 0x40d97c8d 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_SuppSvcFrameworkCore.c] - [MAIN]_SuppSvcReleaseActiveList
[LteRrc] 0x414db999 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]CLEAR OPFLAG :(MAIN-SUPP_SVC_CANCLE_DISPATCH)
[LteRrc] 0x414dba49 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][ProcessMsg] LTERRC_MSG_EMPTY
[LteRrc] 0x40d5aebb 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_ProcSysInfo.c] - [MAIN]NOT in Background operation (M_state: 1)
[LteRrc] 0x414dd8df 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]DrxEnd: gDrxRrc_Flag 0 gDrxL1_Flag 1 gDrxRrc_SaveL1Flag 1
[LteRrc] 0x414dd917 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Support Band num(0) band1(0) band2(0) ALPSS REL(2) Capa REL(4)
[LteRrc] 0x414dd937 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Entity (83) (19) (201)
```
It doesn't look like the message has touched a decoder. Can you give a concrete example of a queue item for any RRC messages?

mariusmue, 2022-03-18:
Unfortunately, it's a bit more complex in this scenario. There are two important things to consider here
1) The op for the message can change across baseband versions. To automate and ease our live, we hence included `SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID` in PatternDB.
2) The payload structure for baseband internal messages to the LTE_RRC task changes depending on the used op. For fuzzing, we used `LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID` - unfortunately, the payload for messages of this type behave differently from what `send_rrc` expect.

We never really injected `LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID` from guestlink, but I did some experiment and the things below should work. This is following the message description as described in the source code for the fuzz task.

```Python
# change logging to only include relevant parts
self.guest_logger.task_log_disable_all()
self.guest_logger.task_log_enable('LteRrc')

# our variables
asn_pl = b"\x20\x1b\x3f\x80\x00\x00\x00\x01\xa9\x08\x80\x00\x00\x29\x00\x97\x80\x00\x00\x00\x01\x04\x22\x14\x00\xf8\x02\x0a\xc0\x60\x00\xa0\x0c\x80\x42\x02\x9f\x43\x07\xda\xbc\xf8\x4b\x32\x18\x34\xc0\x00\x2d\x68\x08\x5e\x18\x00\x16\x80\x00"
unused = 0
pdu = 0 # You will need to change this. Either static baseband RE, or trying and checking FirmWire's output
op = self.loader.symbol_table.lookup('SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID').address

# create clean working state
self.restore_snapshot('interactive') 
gl = self.get_peripheral('glink')

# we will need a allocated chunk in memory to hold the ASN payload
gl.create_block(len(asn_pl))
self.run_for(1)
block_addr = gl.access
self.qemu.wm(block_addr, 1,  asn_pl, raw=True)

# Create message as described in fuzz task header
pl = struct.pack('<IIII', unused, pdu, len(asn_pl), block_addr)

# Send the message in the right format (which is, a "direct" message whose pl is UNUSED+PDU+LEN+*ASN_PL)
gl.send_queue_op(False, 'LTERRC', op, 0, pl)
gl.set_event('LTE_RRC_') # LTE RRC messages need to have an event set
self.run_for(1)
```

Running above code from the console results, on my end, to the following output:
```
[2491.02450] 575 total log lines omitted [Background=104 INTERACTIVE=104 MTI=101 HISR2=54 CDMOT=50 BTL=40 DS_DBG_SAP=30 RLC=29 LTE_TCPIP=24 DBG_SAP=21 ...]
[2491.02462][LteRrc] pal_TaskEntry_LteRrc+0x91 (0x408c193b) 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]DrxStart: gDrxRrc_Flag 0 gDrxL1_Flag 1 gDrxRrc_SaveL1Flag 1
[2491.02569][LteRrc] LteRrc_ProcRxMsgFn+0x153 (0x40e648e7) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_CommDb.c] - [MAIN][LTERRC_INT_MOB_CMD_HO_FROM_IRAT] RegAllocList
[2491.02677][LteRrc] 0x408bf785 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][RxMsg SelectMsgQ] Select LteRrc_CurMsgQ
[2491.02737][LteRrc] LteRrc_ReceiveMsg+0x8b9 (0x408c04e9) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][RxMsg GetMsgDesc] LTERRC_RADIO_MSG_TYPE:: No MsgDesc
[2491.02796][LteRrc] LteRrc_DisplayRxMsg+0x303 (0x408bd01f) 0b11: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][RX][DO] 1. [LTERRC] <== LTERRC_INT_MOB_CMD_HO_FROM_IRAT (0xc3a0)[Init][Wait Msg]
[2491.02859][LteRrc] LteRrcDsds_CheckIsProcStart+0xa9 (0x40904b91) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_ProcDsds.c] - [MAIN][LTE RRC DSRC] LteRrcDsds_CheckIsProcStart msgtype(4)
[2491.02919][LteRrc] LteRrcProAsnDecode+0x4f (0x40827b81) 0b101: [../../../../../../CALPSS/LteL3/LteRrc/asn/arm/Code/Rel1510/src/LteRrc_Codec.c] - LteRrcProAsnDecode (pdu: 6)
[2491.02963][LteRrc] AsnMemAlloc+0x5d (0x411e8a8d) 0b10: [../../../PALCommon/C-Common/StackCommon/Common/Code/src/smpfp_AsnCodecInterface.c] - PROASN_MALLOC Addr:0x4439C020 PDU:6 size:1532
[2491.03007][LteRrc] TraceAsnDecode+0x3b (0x411e8c49) 0b10: [../../../PALCommon/C-Common/StackCommon/Common/Code/src/smpfp_AsnCodecInterface.c] - Output(0x0) from Buffer(0x20) with unpackingLen(1)/unpackedLen(1) @line 121912 (BitUnpacking8)
[2491.03033][LteRrc] TraceAsnDecode+0x3b (0x411e8c49) 0b10: [../../../PALCommon/C-Common/StackCommon/Common/Code/src/smpfp_AsnCodecInterface.c] - Output(0x1) from Buffer(0x1B) with unpackingLen(2)/unpackedLen(3) @line 121919 (BitUnpacking8)
[2491.03058][LteRrc] TraceAsnDecode+0x3b (0x411e8c49) 0b10: [../../../PALCommon/C-Common/StackCommon/Common/Code/src/smpfp_AsnCodecInterface.c] - Output(0x0) from Buffer(0x1B) with unpackingLen(1)/unpackedLen(4) @line 25854 (BitUnpacking8)
[2491.03083][LteRrc] TraceAsnDecode+0x3b (0x411e8c49) 0b10: [../../../PALCommon/C-Common/StackCommon/Common/Code/src/smpfp_AsnCodecInterface.c] - Output(0x0) from Buffer(0x1B) with unpackingLen(1)/unpackedLen(5) @line 18270 (BitUnpacking8)
[2491.03119][LteRrc] LteRrcAsn_DecodeMsg+0xbd (0x40d7d2cb) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_CommUtil.c] - [MAIN]LteRrc_ProAsnDecode: Success
[2491.03155][LteRrc] LteRrcAsn_DecodeMsg+0x1d9 (0x40d7d3e7) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_CommUtil.c] - [MAIN]LteRrc_ProAsnDecode Success, pdu(6) unpackedLen(5 bits)
[2491.03189][LteRrc] LteRrc_ProcessDlAsn1Msg+0xe9 (0x408c0b6d) 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN][ProAsn] LteRrc_ProcessDlAsn1Msg (DecodedData: 0x4439c020)
[2491.03242][LteRrc] LteRrcUtil_SaveDmCurRadioMsg+0x89 (0x40d7f8f1) 0b101: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_CommUtil.c] - [MAIN]LteRrcUtil_SaveDmCurRadioMsg direction (2)
[... cut ...]
[2491.05042][LteRrc] pal_TaskEntry_LteRrc+0xd23 (0x408c25cd) 0b100: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]DrxEnd: gDrxRrc_Flag 0 gDrxL1_Flag 1 gDrxRrc_SaveL1Flag 1
[2491.05073][LteRrc] pal_TaskEntry_LteRrc+0xd61 (0x408c260b) 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Support Band num(0) band1(0) band2(0) ALPSS REL(2) Capa REL(4)
[2491.05102][LteRrc] pal_TaskEntry_LteRrc+0xd8d (0x408c2637) 0b0: [../../../CALPSS/LteL3/LteRrc/Code/src/LteRrc_Task.c] - [MAIN]Entity (88) (19) (201)
```

Note that it attempts ASN decoding, but fails, as the wrong PDU for your payload was set.
Hop this helps!

Elkamika, 2022-12-06:
Hi, @mariusmue @grant-h , I have the following code and console output. 
Why don't I see the different IE inside the RRC message in the console ? or is it being processed in the background?
The RRCConnectionReconfiguration message seems to be decoded successfully but doesn't show more information about it. Did I miss something?

#include <common.h>
#include <shannon.h>
#include <modkit.h>
#include <task.h>

char buf[] = "\x22\x12\x15\xe8\x00\x04\x05\x84\xc9\x00\x41\x6e\x1a\x6d\xc4\x0f"
"\x88\xc2\x11\x21\x4a\x25\xdb\x44\xdb\xa8\x9f\x69\x14\x47\x22\x9c" 
"\xd4\x55\x7e\x8a\xc6\xd1\x64\x6d\xad\xb6\x1d\xd6\xb3\xbe\x0c\x78" 
"\x53\xce\x30\x02\x90\x28\x79\xf0\x84\x18\x1a\x04\x3c\xf8\x80\x87" 
"\x81\xcf\x80\x60\x45\x41\x27\xc0\x30\x00\x01\x00\x46\x01\x18\x03" 
"\x44\x84\x6b\x62\x6a\xd4\xe0\x9f\xec\xa0\x20\xfa\x8d\xac\x31\x8f" 
"\xac\x50\x72\x4f\x3a\x21\xba\xec\x8d\x24\xf4\x84\x07\x6c\xe0\x88"
"\x82\x00\x3c\x26\x06\x32\x86";

const char TASK_NAME[] = "AFL_LTE_RRC\0";
struct qitem_lte_rrc {
  struct qitem_header header;
  uint32_t unused;
  uint32_t pdu_type;
  uint32_t pl_size;
  char * asn_pl;
} PACKED;

static uint32_t qid;
static struct pal_event_group * group;

void task_main()
{

    qid = queuename2id("LTERRC");
    group = eventname2addr("LTE_RRC_");

    uint32_t input_size;
    uint16_t size;
    uart_puts("[+] Allocating Qitem\n");

    struct qitem_lte_rrc * item = pal_MemAlloc(4, sizeof(struct qitem_lte_rrc), __FILE__, __LINE__);
    if (!item) {
      uart_puts("ALLOC FAILED");
      return;
    }

    char * pkt = buf;
    input_size = sizeof(buf);
   

    char * asn_pl = pal_MemAlloc(4, input_size, __FILE__, __LINE__);

    uart_puts("[+] Filling the qitem\n");
    item->header.msgGroup = 0;
    item->header.size = sizeof(struct qitem_lte_rrc) - sizeof(struct qitem_header);
    item->header.op = SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID;
    item->unused = 0x00;
    item->pdu_type = 1;
    item->pl_size = (input_size); 

    memcpy(asn_pl, pkt, input_size);
    item->asn_pl = asn_pl;

 
    pal_MsgSendTo(qid, item, 2);

    pal_SmSetEvent(&group, 4);
}

Console output 

[image]
The IEs inside this RRC message 
[image]

helpcomputer1999, 2022-12-21:
> 

How do you access the firmware DBG messages while running in Console mode?  The documentation does not make this clear.