FirmWire/FirmWire #3: MTK NV Data Compatability

j4s0n, 2022-03-25:
When running FirmWire against any MTK images, the `--mtk-loader-nv_data` parameter is required (or `./mnt/vendor/nvdata` needs to exist). If a path is supplied (or `mkdir -p mnt/vendor/nvdata`), then FirmWire crashes when trying to create the NVRAM subdirectory:

```
[INFO] firmwire.vendor.mtk.machine: Activating SCPCCISM via affinity hook
[1.59030][NO_TASK] 0x90278d4d [CCISMC] =========> ccismc_create  [CCISM_TR_TASK_CREATE]
[1.59286][NO_TASK] 0x90278d4d [EMM RATCHG] >> CEmmRatChg::CEmmRatChg() [EMM_RATCHG_FUNC_constructor]
[1.82042][NO_TASK] 0x90278d4d [CCCI_FS] ===> MD_FS_GetDiskInfo  [CCCIFS_TR_GETDKINFO_IN]
[1.82046] last message matched ban pattern 'CCCI_FS'
[1.82093][NO_TASK] 0x90151a1b [CCCI_FS GetDiskInfo] filename: Z:\
[1.82094] last message matched ban pattern 'CCCI_FS'
[1.93383] 12 total log lines omitted [NO_TASK=12]
[1.93390][NO_TASK] 0x90150fd7 [CCCI_FS GetAttributes] filename: Z:\FAT2E4C5231.log
[1.93391] last message matched ban pattern 'CCCI_FS'
[1.97289] 11 total log lines omitted [NO_TASK=11]
[1.97296][NO_TASK] 0x901509f7 [CCCI_FS Open] filename: Z:\NVRAM
[1.97297] last message matched ban pattern 'CCCI_FS'
[1.98623] 10 total log lines omitted [NO_TASK=10]
[1.98631][NO_TASK] 0x90279111 nvram_init, step=6213dddc, para=62109724 [FUNC_NVRAM_INIT]
[ERROR] firmwire.hw.peripheral.SHM_CCIF_Periph.SharedMemoryCCIF.FSD: Close: invalid handle value 4294967287
[2.09281] 36 total log lines omitted [NO_TASK=36]
[2.09287][NO_TASK] 0x901510b3 [CCCI_FS CreateDir] filename: Z:\NVRAM
[2.09290] last message matched ban pattern 'CCCI_FS'
[ERROR] firmwire.hw.peripheral.SHM_CCIF_Periph.SharedMemoryCCIF.FSD: Unhandled FSD operation 0x1007 (FSCCCIOp.CREATEDIR)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 522, in run
    handler(message)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/watchmen.py", line 78, in watchtrigger
    ret = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 491, in _handle_remote_memory_write_message
    success = mem_range.forwarded_to.write_memory(
  File "/usr/local/lib/python3.8/dist-packages/avatar2/peripherals/avatar_peripheral.py", line 66, in write_memory
    return intervals.pop().data(offset, size, value, **kwargs)
  File "/root/FirmWire/firmwire/vendor/mtk/hw/PCCIFPeripheral.py", line 500, in hw_write
    self.handleFSPacket(ring, packet)
  File "/root/FirmWire/firmwire/vendor/mtk/hw/PCCIFPeripheral.py", line 605, in _handleFSPacketEmu
    emu_resp = ring.parent.fsd_ctx["emu"].handle_packet(
  File "/root/FirmWire/firmwire/vendor/mtk/hw/FSD.py", line 573, in handle_packet
    resp_packs = self._handle_op(op, packs)
  File "/root/FirmWire/firmwire/vendor/mtk/hw/FSD.py", line 630, in _handle_op
    assert 0
AssertionError
```

Adding support for the `CREATEDIR` operation is trivial. Doing so causes the NV initialization routine to execute, which seeds the NV data. This subsequently allows the baseband to fully boot to the `[SSIPC][ILM_MSG] waiting msg` loop state. While this may not be ideal for all circumstances, it removes a major hurdle to analyze new images without seeding with arbitrary NV data snapshot.

Apart from allowing the init procedures to auto-generate NV data, what's the recommended method for providing seed NV data? Extract from the AP archive, or from a running device? Any idea on the scope of applicability of NV data extracted from one FW/device different FW builds, builds for different devices, or different chipsets? Unless relying on CREATEDIR seems optimal, I suggest adding some clarification in docs, as it will be helpful to others! If CREATEDIR is optimal, then it make be preferable to just automatically create `./mnt/vendor/nvdata` when it is needed, but not present.

I will issue a pull request against this Issue to add support for `CREATEDIR`. In my testing, this allows images to boot cleanly when given an empty NV data directory.

grant-h, 2022-03-28:
> When running FirmWire against any MTK images, the `--mtk-loader-nv_data` ...
> 
> Adding support for the `CREATEDIR` operation is trivial. Doing so causes the NV initialization routine to execute, which seeds the NV data. This subsequently allows the baseband to fully boot to the `[SSIPC][ILM_MSG] waiting msg` loop state. While this may not be ideal for all circumstances, it removes a major hurdle to analyze new images without seeding with arbitrary NV data snapshot.

Thank you for your contribution to FirmWire! 

> Apart from allowing the init procedures to auto-generate NV data, what's the recommended method for providing seed NV data? Extract from the AP archive, or from a running device?

All of our extractions were from running devices. I have not explored extracting from AP archives and I don't believe this would work since NV data is factory provisioned (IMEI, etc.). It's not clear if there would be limitations on the firmware after an NV auto-init as well.

> Any idea on the scope of applicability of NV data extracted from one FW/device to different FW builds, builds for different devices, or different chipsets?

I don't have something to reference, but I believe that since basebands undergo OTA updates, they would need to smoothly transition NV items between versions and create missing ones with default values.

> Unless relying on CREATEDIR seems optimal, I suggest adding some clarification in docs, as it will be helpful to others! If CREATEDIR is optimal, then it make be preferable to just automatically create `./mnt/vendor/nvdata` when it is needed, but not present.

We can clarify the docs (how to do an extraction, default behavior) and add some more warnings about NV data being automatically created.

> I will issue a pull request against this Issue to add support for `CREATEDIR`. In my testing, this allows images to boot cleanly when given an empty NV data directory.

Please do so we can test with some more firmware.

j4s0n, 2022-03-29:
> Please do so we can test with some more firmware.

Done! https://github.com/FirmWire/FirmWire/pull/4

It would be interesting to compare fuzzing coverage when using the init NV data vs a dataset extracted from a live device. In my simple boot-up testing, the log output suggests that the differences are minimal.

grant-h, 2022-03-29:
@j4s0n I am unable to reproduce CREATEDIR being called. I used the `A415FXXU1ATE1_CP15883562_CL18317596_QB31188168_REV00` img and it just hangs trying to restore NVD_IMEI

```
[WARN] firmwire.hw.peripheral.SHM_CCIF_Periph.SharedMemoryCCIF.FSD: Restore: mtk_images/A415FXXU1ATE1_CP15883562_CL18317596_QB31188168_REV00/md1img.img_workspace/ccci_nvdatacfqdwhfi/mnt/vendor/nvdata/md/NVRAM/NVD_IMEI/CS17_000 stub!
```

What firmware version are you looking at?

> It would be interesting to compare fuzzing coverage when using the init NV data vs a dataset extracted from a live device. In my simple boot-up testing, the log output suggests that the differences are minimal.

NV data could be used to gate major features so it's possible big changes could be seen.

j4s0n, 2022-03-29:
I've observed this behavior as well, but only against an unsupported chipset (P35) -- I assumed that was the cause. I see that the RESTORE op has been stubbed out due to an issue in testing with A315GXXU1ATG1. Could stubbing this out be the reason for the loop?

I've been testing against an SM-A107M (A10S -- P22) firmware version A107MUBU5CUI1.

To reproduce:
```
rm -rf ./mnt
rm -rf ~/md1img_a10s.img_workspace/
mkdir -p ./mnt/vendor/nvdata
./firmwire.py ~/md1img_a10s.img
```

See attached log
[A10S_CREATEDIR.log](https://github.com/FirmWire/FirmWire/files/8374468/A10S_CREATEDIR.log)

Without this patch (and with no seeded NVDATA), FirmWire crashes.

grant-h, 2022-04-01:
Thank you for your contribution!

commit 149cc93d (2022-04-01): MTK: FSD add CREATEDIR op (#4)

Closes #3
closed 2022-04-01 by grant-h