FirmWire/FirmWire #5: Best practice of taking snapshots

MustBastani, 2022-04-04:
What is a good address to take a snapshot at to keep the state of the baseband before calling the guest link methods?

I used the address of the INTERACTIVE task in two scenarios, and both failed.
First:
```bash
./firmwire.py -t glink --console -S modem.bin
```
In jupyter console: 
```bash
self.snapshot_state_at_address(0x4b000001, 'interactive')
# Send a message with glink.send_queue_op
self.restore_snapshot('interactive')
# Get segmentation fault in the first terminal
```

Second:
```bash
./firmwire.py -t glink --console -S modem.bin
```
In jupyter console (after enough time):
```bash
self.qemu.stop() 
self.snapshot('interactive')
# Send a message with glink.send_queue_op and see logs from the target task.
self.restore_snapshot('interactive')
# Repeat the previous message with glink.send_queue_op and don't see any logs from the target task.
```

Test [modem.bin](https://github.com/grant-h/ShannonFirmware/blob/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4)

mariusmue, 2022-04-05:
The second one is the way how I use snapshots and usually don't have any issues.

However, one caveat: After restoring a snapshot, you need to get a new reference to glink before calling `send_queue_op`.
So, please add the line:
```python
glink = self.get_peripheral("glink")
```
after snapshot restore, and check if the issue still persist.

(Lastly, there is a third method for snapshot creation, by using the `--snapshot-at` CLI argument)

grant-h, 2022-04-05:
You mention that the target segfaults. This is a known issue that can occur after taking a snapshot and the target continues to run. The snapshot will take properly, but panda will segfault afterwards for a difficult to debug reason.

The second issue was mentioned by Marius - all peripherals are replaced with new objects on snapshot restore, meaning you need to get a new reference to them after each time.

closed 2022-06-07 by grant-h