FirmWire/FirmWire #6: Missing functionality

azazqi, 2022-04-11:
In the README.md, it says:

> Upon a vendor's request, the current public release of FirmWire is a preview version omitting some of the functionality described in the paper. We will publish the full version and automated scripts to replicate our experiments during NDSS'22 (April 24th-28th).

Which functionality is currently missing, that will be published near the end of April?

mariusmue, 2022-04-13:
Hi.
We cannot comment on this for the time being, but the release will be in approximately two weeks, so we hope the last bit of waiting won't be too annoying. :)

lordofthefarts, 2022-05-05:
hello - is there an update on this please? interested to experiment but was holding out on this

mariusmue, 2022-05-08:
Heya, thanks for your interest in FirmWire! We are working hard besides the scene to make the second stage of the release available to the public.

By now, we released all parts of FirmWire itself in this very repository, and we hope that we can push our experiment code in the next couple of days (it will live in a seperate repository). We also uploaded/archived our large-scale analysis data-set already over here: https://zenodo.org/record/6516030#.YncQV3VByEI

Either way, you can already go and explore FirmWire as of now, all remaining additions will just be for reproducing the experiments reported in the paper!

lordofthefarts, 2022-05-09:
thanks for the update mariusmue - great work !

closed 2022-09-19 by grant-h

nishant412, 2023-10-16:
Hello all, this emulation platform is amazing! Am starting to look into this, and was wondering if there are plans to release scripts to reproduce some of the vulns from the paper? In particular was interested in reproducing the ground truth vuln that you folks showed (SM1)

grant-h, 2023-10-16:
> Hello all, this emulation platform looks amazing! Am starting to look into this, and was wondering if there are plans to release scripts to reproduce some of the vulns from the paper? In particular was interested in reproducing the ground truth vuln that you folks showed (SM1)

Check out
https://github.com/FirmWire/ndss22_experiments

nishant412, 2023-10-19:
Thank you so much, this is really helpful!!