FirmWire/FirmWire #10: Fatal error FRPKSIG - Dev Assert DSP PSQ overflow

lordofthefarts, 2022-07-08:
hello, 

I have been running though your Quick Start and grabbed a Shannon firmware to emulate (CP_G930FXXS5ESF1_CP12893112_CL14843133_QB24085562_REV00_user_low_ship.tar.md5) this was taken from the data set (https://zenodo.org/record/6516030#.YncQV3VByEI) kindly  provided on ticket #6 by @mariusmue; I think this was the first Shannon baseband on this list. 

I have had success emulating other Shannon basebands but this one causes a FATAL ERROR due to (reason:Dev Assert DSP PSQ overflow) when running under normal emulation, I have attached a log of the output that dumps stack and register state. To understand more about your tool, I attempted to debug this problem dynamically using GDB watchpoints, however this is causing an exception in `/avatar2/plugins/gdbserver.py` (I should probably raise a separate issue here?)

My question is - have you seen this DSP PSQ overflow as a common issue whilst developing support for Shannon firmwares? If so can you suggest an approach to fix the crash?

[firmwire_log.0.txt](https://github.com/FirmWire/FirmWire/files/9070944/firmwire_log.0.txt)

mariusmue, 2022-07-10:
Hi!

The uploaded firmware images in the data set are meant to replicate our results from the paper - as such, they are indeed not supposed to work all out of the box. We are still in the process of preparing the raw data of these experiments to a separated repository. During the large-scale study, we did not look into the individual bugs/issues - but instead wanted to provide insight on the images were FirmWire runs out of the box without additional modification.

It's likely that with a couple of fixes, these errors are gone, but I don't have a solution ready out of the top of my head without investigating this issue a bit more.

For more context, here is the relevant Figure from the paper: As visible, FirmWire has better support for G950+ firmware.

[image]

lordofthefarts, 2022-07-14:
thanks for the reply; I have read the paper a while back so forgive me for forgetting this table ! I will look into the crashing on S8 and see if I can figure out what might be causing it.

closed 2022-07-14 by lordofthefarts