FirmWire/FirmWire #11: Unexpectedly exited when running the fuzz demo code on Shannon Baseband

N3vv, 2022-08-09:
Hello, When I run demo code of fuzzing on Shannon baseband, it seems to have just dropped out. The Shannon baseband firmware  is downloaded from the test set given in the paper with version CP_G973FXXS3ASJA_CP14156780_CL17063867_QB26713219_REV01.

-------------
# AFL_FORKSRV_INIT_TMOUT=10000 AFL_COMPCOV_LEVEL=1 AFL_DEBUG=1 ./AFLplusplus-stable/afl-fuzz -i in -o out -U -- ./firmwire.py --restore-snapshot fuzz_base --fuzz gsm_cc --fuzz-input @@ new_modem.bin
.....
[INFO] firmwire.vendor.shannon.machine: TASK245: hGmcTime [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK246: hWakeUpD [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK247: AFL_GSM_CC (0x4b000001)
[INFO] firmwire: Starting emulator ShannonEMU3334
==> BOOT
Got AFL_COMPCOV_LEVEL 1.
==> WAIT SHUTDOWN

[-] PROGRAM ABORT : Timeout while initializing fork server (setting AFL_FORKSRV_INIT_TMOUT may help)
         Location : afl_fsrv_start(), src/afl-forkserver.c:1033

I try to set AFL_FORKSRV_INIT_TMOUT to a large value but it still fails.I would appreciate it if you could give me some helpful advice.

domenukk, 2022-08-09:
How did you create the snapshot?

N3vv, 2022-08-09:
First, I used the command `python3 firmwire. Py new_modem.bin` to get the snapshot-at address. Then executed the command `python3 firmwire.py new_modem.bin --snapshot-at 0x4054f709,fuzz_base`:
```
# python3 firmwire. Py new_modem.bin
...
[83.84793][Background] 0x407dd87b 0b101: [../../../VARIANT/PALVar/C-Var/NVSS/VcgMsg/src/vcg_Msg.c] - No Active Calls!!!
[83.84900][TaskReg] 0x40557d1b pal_Sleep(5000000)
[83.84997][LTE_DM] 0x40bd74cf 0b1: [../../../LTESAE/LHAL/Common/src/hal_drx.c] - [DRX]gDrx_ActiveRat_[0][1]=0,0 (0)
[83.85129][BTL] pal_SmSetEvent+0x9e1 (0x4054f04b) 0b10: [../../../VARIANT/PALVar/Platform_EV/PAL/BackTraceLog/src/pal_BackTraceLog.c] - [BTL] btlExtLogProcTimerHandler called
[83.85174][BTL] 0x4054f6c5 0b10: [../../../VARIANT/PALVar/Platform_EV/PAL/BackTraceLog/src/pal_BackTraceLog.c] - [BTL] bltProcessExternalLog
[83.85199][BTL] 0x4054f6f3 0b10: [../../../VARIANT/PALVar/Platform_EV/PAL/BackTraceLog/src/pal_BackTraceLog.c] - [BTL] ABOX log = 0
[83.85218][BTL] 0x4054f709 0b10: [../../../VARIANT/PALVar/Platform_EV/PAL/BackTraceLog/src/pal_BackTraceLog.c] - [BTL] L-CPU log = 0
... loop ...
# python3 firmwire.py new_modem.bin --snapshot-at 0x4054f709,fuzz_base
```

mariusmue, 2022-08-09:
Hi. When creating the snapshot, you most likely already want to have the fuzz-task injected!
Hence, the commandline should look something like:
`python3 firmwire.py new_modem.bin --snapshot-at 0x4054f709,fuzz_base --fuzz gsm_cc --fuzz-input /tmp/a`

Besides this, for getting more output during debugging, you can also run the afl command line with the `AFL_DEBUG_CHILD=1` environment variable and replace `--fuzz` with `--fuzz-triage`.

Hope that helps!

N3vv, 2022-08-10:
As you suggested, I didn't generate snapshots in the right way before.  By the way, using `--fuzz-triage` is really helpful. I think this issue can be closed now. Many thanks for your patient reply!!

closed 2022-08-10 by domenukk