FirmWire/FirmWire #16: Fuzzing pal_MemFree symbol not found: Examples fuzzing?

Michel-de-Boer-dev, 2022-09-30:
Hello!

I am trying to reproduce some of the paper fuzzing results. Unfortunately I have not much luck with getting any fuzzing to work.
 
AFL++ version v3.13C
Running FirmWire inside Docker container.

Using the following command:
`AFL_DEBUG=1 afl-fuzz -i in -o out -U -- ./firmwire.py --fuzz gsm_cc --fuzz-input @@ ShannonFirmware/modem_files/CP_G973FXXU9FUCD_CP18513696_CL21324211_QB39036441_REV01_user_low_ship.tar.md5.lz4
`
I get:
`
[ERROR] firmwire.vendor.shannon.machine: Unable to resolve requested dynamic modkit symbol pal_MemFree
[ERROR] firmwire.vendor.shannon.machine: Failed to inject task into OS
`

When checking, it indeed seems like the firmware itself is missing these symbols.  I got the firmware from https://github.com/grant-h/ShannonFirmware.

Removing the registration in shanon.h works, but breaks the emulation. (Setting AFL_FORKSRV_INIT_TMOUT very high does not help, it hangs at `[INFO] firmwire.hw.peripheral.ShannonSOCPeripheral.SOC: CHIP_ID read: 50000000)`

What are the commands used in the paper? Any help would be appreciated.

Thank you very much!

mariusmue, 2022-09-30:
Hi,

Thanks for reporting. The modem files themselves don't have any symbols, we use patternDB to reconstruct them.
pal_memFree was added after the initial release, together with some changes in the injected tasks. I just pushed 2d9f4c503d1caa5f021be167273e354b1071bab1, which should fix this.

Locally, when I run `python3 -u ./firmwire.py ./modem_files/CP_G973FXXU9FUCD_CP18513696_CL21324211_QB39036441_REV01_user_low_ship.tar.md5.lz4 --fuzz-triage gsm_cc --fuzz-input /tmp/a` the fuzzing input is taken and executed as expected.

Regarding the commands in the paper: Unfortunately, we still need to hold back from publishing the experiment code, which is not by our choice. However, the command line we used for fuzzing was:
```
$ AFL_NO_UI=1 AFL_FORKSRV_INIT_TMOUT=100000 \
  timeout 86400 "$EXPERIMENT_ROOT/AFLplusplus/afl-fuzz" \
  -i in  -o out -t 10000 -m none -M "main" -U -- ./firmwire.py --fuzz gsm_cc --fuzz-input @@  --consecutive-ports [port_base] --restore-snapshot [snapshot] [image]
```

jeppojeps, 2022-10-18:
it seems afl-fuzz works only when --fuzz-triage is used, weirdly, but with --fuzz it remans stuck for mor than the TMOUT=100000 and dies

mariusmue, 2022-10-20:
Did you start fuzzing from a snapshot, as in my CLI example above, or just trying to run it from the very beginning?
In fuzz mode, we disconnect most of the avatar orchestration, which can result into problems during the bootup of some images.

mariusmue, 2022-11-04:
Similar Issue was in #19, with the following solution: https://github.com/FirmWire/FirmWire/issues/19#issuecomment-1297414972

Hence, closing this issue for now. If the problem still exists after following the instructions above, feel free to reopen/post below.

closed 2022-11-04 by mariusmue