FirmWire/FirmWire #19: Libpanda always starts in suspended mode, even in Fuzz mode

jeppojeps, 2022-10-20:
Hello, 

I don't really understand how this is even possible if I comment out the gdb ports/commands in the different Python files, I have also been checking the version of panda and pypanda, no matter what the only thing I can alter are the port numbers, where exactly the command line is prepared? I always get 

`[PYPANDA] Panda args: [/usr/local/lib/python3.8/dist-packages/pandare/data/arm-softmmu/libpanda-arm.so -L /usr/local/lib/python3.8/dist-packages/pandare/data/pc-bios -machine configurable -kernel CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4_workspace/ShannonEMU_conf.json -gdb tcp::3355 -S -drive if=none,id=drive0,file=CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4_workspace/snapshots.qcow2,format=qcow2 -nographic -qmp tcp:127.0.0.1:3335,server,nowait -m 128M -monitor unix:/tmp/pypanda_mc5sbarnc,server,nowait]`

hence AFL does not fire cause the forkserver does not start, and if I plug gdb and continue all goes bananas.

mariusmue, 2022-10-20:
Hi,
Please *do not* comment the GDB interaction for pypanda. Pypanda/libpanda is orchestrated via avatar2, which expects it to be stopped at startup (hence the '-S' in the commandline you see).

If you supply `-S` on the FirmWire commandline, avatar2 will leave pypanda in the stopped state - otherwise, it just automatically starts it via the internal `.cont()` function.
If you want to attach gdb yourself, use the according CLI argument; avatar2 will then open a separate gdb server to which you can connect.

The forkserver should be independent of all of this.

jeppojeps, 2022-10-20:
Hi Marius, thanks for the prompt response. I saw your cont() and also the default start_suspended=False, but I think I am missing something. I am assuming some sequentiality, like I start the command above, all fires up, the cont() hits the GSM task starts along with the forkserver. What happens to the above execution is that it timeouts w/o starting the firmware, as mentioned the machine seems to remain suspended until AFL timeouts. Now, my checks above were just an attempt to see if something was going wrong. I can't really understand why it remains suspended and never recovers, I have also put the AFL_TIMEOUT to a gigantic value but it doesn't seem to move on from a suspended start. Likely I am doing something wrong. But I have also tried as you mentioned in another issue, to execute a fuzz-triage with a specific payload and in that case it seems that the payload doesn't go through the task, because no prints i.e. FIRE! are shown whatsoever. 

Thanks for your patience.

mariusmue, 2022-10-20:
Out of the blue, it's difficult to help, as it could be a variety of reasons.
- did you inject the fuzz task?
- did you try running AFL++ with `AFL_DEBUG_CHILD` env variable set to see child output
- did you try running `fuzz-triage` rather than fuzz to leave output enabled
- are you starting fuzzing from a snapshot after boot?

There are a lot of different things, but I'm pretty sure that fuzzing should work on an untouched code base with known-to-work firmware images.

jeppojeps, 2022-10-20:
I am not starting from a snapshot and I haven't tried the AFL_DEBUG_CHILD, I think I probably got the point, the snapshot is likely important, I was assuming, wrongly, that I could attach the fuzzer on the first boot. Will let you know. Thanks again.

jeppojeps, 2022-10-21:
Hello,

after being able to get a snapshot through the jupiter cosole, the command like seems weirdly broken, I set the variables you suggested and got pretty much the same output. I am assuming the execution is just stuck somewhere.

`root@5ccca7153214:/firmwire# export AFL_DEBUG_CHILD=1
root@5ccca7153214:/firmwire# export AFL_FORKSRV_INIT_TMOUT=100000
root@5ccca7153214:/firmwire# afl-fuzz -i fwdocker/inputsFW/ -o out -U -- ./firmwire.py --restore-snapshot tst --fuzz gsm_cc --fuzz-input @@ CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4
[+] Loaded environment variable AFL_SKIP_CPUFREQ with value 1
[+] Loaded environment variable AFL_DEBUG_CHILD with value 1
[+] Loaded environment variable AFL_FORKSRV_INIT_TMOUT with value 100000
afl-fuzz++3.13c based on afl by Michal Zalewski and a large online community
[+] afl++ is maintained by Marc "van Hauser" Heuse, Heiko "hexcoder" Eißfeldt, Andrea Fioraldi and Dominik Maier
[+] afl++ is open source, get it at https://github.com/AFLplusplus/AFLplusplus
[+] NOTE: This is v3.x which changes defaults and behaviours - see README.md
[+] No -M/-S set, autoconfiguring for "-S default"
[*] Getting to work...
[+] Using exponential power schedule (FAST)
[+] Enabled testcache with 50 MB
[*] Checking core_pattern...
[+] You have 20 CPU cores and 3 runnable tasks (utilization: 15%).
[+] Try parallel jobs - see /usr/local/share/doc/afl/parallel_fuzzing.md.
[*] Setting up output directories...
[+] Output directory exists but deemed OK to reuse.
[*] Deleting old session data...
[+] Output dir cleanup successful.
[*] Checking CPU core loadout...
[+] Found a free CPU core, try binding to #0.
[*] Scanning 'fwdocker/inputsFW/'...
[+] Loaded a total of 3 seeds.
[*] Creating hard links for all input files...
[*] Validating target binary...
[*] No auto-generated dictionary tokens to reuse.
[*] Attempting dry run with 'id:000000,time:0,orig:p1'...
[*] Spinning up the fork server...
              ___            __      _                          
-.     .-.   | __|(+) _ _ _ _\ \    / /(+) _ _ ___    .-.     .-
  \   /   \  | _|  | | '_| '  \ \/\/ /  | | '_/ -_)  /   \   /  
   '-'     '-|_|   | |_| |_|_|_\_/\_/   | |_| \___|-'     '-'   
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~             
                A  baseband  analysis  platform
                   https://github.com/FirmWire

[INFO] firmwire.loader: Reading firmware using ShannonLoader (shannon)
[INFO] firmwire.vendor.shannon.loader: SoC <ShannonSOC S5000AP - 20190103> (automatic)
[INFO] firmwire.emulator.patterndb: Searching for patterns in [40010000 - 425579a0]
[INFO] firmwire.emulator.patterndb: Found symbol boot_mpu_table -> 41777400 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_setup_memory -> 40415088 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_key_check -> 40549f8a [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol OS_fatal_error -> 4054ccb2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemAlloc -> 423916b2 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Fixing up TCM region symbol pal_MemAlloc (423916b2 -> 0401581e)
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemFree -> 40cbb1ac [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MsgSendTo -> 41155e80 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_Sleep -> 40cba1f8 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol log_printf -> 405489ae [CACHED]
[WARN] firmwire.emulator.patterndb: Unable to resolve dynamic symbol log_printf2. Functionality may be affected
[INFO] firmwire.emulator.patterndb: Found symbol pal_SmSetEvent -> 4054d5a2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_EVENT_GROUP_LIST -> 416f93ec [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Dereference [0x416f93ec] -> 0x418385e8
[INFO] firmwire.emulator.patterndb: Found symbol SYM_TASK_LIST -> 0176f534 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Found likely task name: b'GMAC\x00\x00\x00\x00UL', keeping task layout
[INFO] firmwire.emulator.patterndb: Found symbol SYM_SCHEDULABLE_TASK_LIST -> 43a36e68 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_CUR_TASK_ID -> 418385f4 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_FN_EXCEPTION_SWITCH -> 40c71734 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_QUEUE_LIST -> 4013d408 [CACHED]
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_SXXXAP_DVFS_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_S337AP_SHM_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Found symbol SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID -> 0000c3a3 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol DSP_SYNC_WORD_0 -> 4060e214 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Retrieved sync word 0: 141
[... 434 lines cut ...]
[INFO] firmwire.vendor.shannon.machine: TASK234: IRATSRCH [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK235: IRATMEAS [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK236: PdcpStat [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK237: PdcpResu [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK238: EMM_TIME [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK239: EMM_TIME [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK240: hUSIM [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK241: hUsimDet [HISR TASK]
[... cut ...]
[INFO] firmwire.vendor.shannon.machine: TASK244: PDA_Acti [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK245: hGmcTime [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK246: hWakeUpD [HISR TASK]
[INFO] firmwire.vendor.shannon.machine: TASK247: AFL_GSM_CC (0x4b000001)
[INFO] firmwire: Starting emulator ShannonEMU3334
==> BOOT
==> WAIT SHUTDOWN
AFL_COMPCOV_LEVEL not set.

[-] PROGRAM ABORT : Timeout while initializing fork server (setting AFL_FORKSRV_INIT_TMOUT may help)
         Location : afl_fsrv_start(), src/afl-forkserver.c:829

`

Thanks

jeppojeps, 2022-10-26:
anyone?

mariusmue, 2022-10-31:
Did you create the snapshot with the fuzz-task already injected? Here is my full commandline log leading to a working fuzzing setup for me:

```
# Run FirmWire once to get snapshot address
$ python3 -u ./firmwire.py ./CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5 

#  search for an output line like this to identify a snapshot address:
# [105.14950][BTL] 0x4054e641 0b10: [../../../VARIANT/PALVar/Platform_EV/PAL/BackTraceLog/src/pal_BackTraceLog.c] - [BTL] L-CPU log = 0

# create dummy test input
$ mkdir in
$ echo AAAA > ./in/a

# create snapshot, with fuzztask injected 
$ python3 -u ./firmwire.py --snapshot-at 0x4054e641,gsm_fuzz_base --fuzz-triage gsm_cc --fuzz-input ./in/a ./CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5 

# ctrl+c because of some locking issue, try again (unfortunately, this happens sometimes)
$ python3 -u ./firmwire.py --snapshot-at 0x4054e641,gsm_fuzz_base --fuzz-triage gsm_cc --fuzz-input ./in/a ./CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5 

# start the fuzzing
$ ./AFLplusplus/afl-fuzz -i ./in -o out -U -- ./firmwire.py --restore-snapshot gsm_fuzz_base --fuzz gsm_cc --fuzz-input @@ CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5   
```

This has been tested with firmwire@2d9f4c503d1caa5f021be167273e354b1071bab1 and afl++@2d640558a09b03e9416b5d87e98cf938b38def9e

jeppojeps, 2022-11-01:
I think my error was to create the snapshot when the firmware is idle. I am going to try again, thanks again for your support.

jeppojeps, 2022-11-04:
Thanks @mariusmue works!

closed 2022-11-04 by mariusmue