FirmWire/FirmWire #20: Cannot connect to Jupiter console

Miracle-doctor, 2022-12-02:
When starting firmware with --console cannot kernel*.json file not created

mariusmue, 2022-12-06:
Hi.

When starting firmwire with --console, it usually enoughs to run `jupyter console --existing`. You are hinting towards an error message encountered during firmwire startup, but without additional logs/infos, it's tough to tell what's going on.

chromed-lucy, 2023-04-12:
I'm also having this issue.

I'm starting firmwire like so
`python3 ./firmwire.py --console --stop https://github.com/grant-h/ShannonFirmware/raw/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4 --debug
              ___            __      _                          
-.     .-.   | __|(+) _ _ _ _\ \    / /(+) _ _ ___    .-.     .-
  \   /   \  | _|  | | '_| '  \ \/\/ /  | | '_/ -_)  /   \   /  
   '-'     '-|_|   | |_| |_|_|_\_/\_/   | |_| \___|-'     '-'   
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~             
                A  baseband  analysis  platform
                   https://github.com/FirmWire

[INFO] firmwire: Downloading https://github.com/grant-h/ShannonFirmware/raw/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4...
[DEBUG] urllib3.connectionpool: Starting new HTTPS connection (1): github.com:443
[DEBUG] urllib3.connectionpool: https://github.com:443 "GET /grant-h/ShannonFirmware/raw/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4 HTTP/1.1" 302 0
[DEBUG] urllib3.connectionpool: Starting new HTTPS connection (1): raw.githubusercontent.com:443
[DEBUG] urllib3.connectionpool: https://raw.githubusercontent.com:443 "GET /grant-h/ShannonFirmware/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4 HTTP/1.1" 200 27040083
[INFO] firmwire: Using cached file CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4
[INFO] firmwire.loader: Reading firmware using ShannonLoader (shannon)
[INFO] firmwire.vendor.shannon.loader: SoC <ShannonSOC S5000AP - 20190103> (automatic)
[INFO] firmwire.emulator.patterndb: Searching for patterns in [40010000 - 425579a0]
[INFO] firmwire.emulator.patterndb: Found symbol boot_mpu_table -> 41777400 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_setup_memory -> 40415088 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_key_check -> 40549f8a [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol OS_fatal_error -> 4054ccb2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemAlloc -> 423916b2 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Fixing up TCM region symbol pal_MemAlloc (423916b2 -> 0401581e)
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemFree -> 40cbb1ac [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MsgSendTo -> 41155e80 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_Sleep -> 40cba1f8 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol log_printf -> 405489ae [CACHED]
[WARN] firmwire.emulator.patterndb: Unable to resolve dynamic symbol log_printf2. Functionality may be affected
[INFO] firmwire.emulator.patterndb: Found symbol pal_SmSetEvent -> 4054d5a2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_EVENT_GROUP_LIST -> 416f93ec [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Dereference [0x416f93ec] -> 0x418385e8
[INFO] firmwire.emulator.patterndb: Found symbol SYM_TASK_LIST -> 0176f534 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Found likely task name: b'GMAC\x00\x00\x00\x00UL', keeping task layout
[INFO] firmwire.emulator.patterndb: Found symbol SYM_SCHEDULABLE_TASK_LIST -> 43a36e68 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_CUR_TASK_ID -> 418385f4 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_FN_EXCEPTION_SWITCH -> 40c71734 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_QUEUE_LIST -> 4013d408 [CACHED]
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_SXXXAP_DVFS_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_S337AP_SHM_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Found symbol SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID -> 0000c3a3 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol DSP_SYNC_WORD_0 -> 4060e214 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Retrieved sync word 0: 141
[INFO] firmwire.emulator.patterndb: Found symbol DSP_SYNC_WORD_1 -> 4060e21a [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Retrieved sync word 1: 286
[INFO] firmwire.emulator.patterndb: Dynamic symbol resolution took 0.02 seconds
[INFO] firmwire.vendor.shannon.loader: Using blank NV data
[INFO] firmwire.loader: Loading complete
[INFO] firmwire: FirmWire initializing ShannonMachine
[INFO] firmwire.emulator.firmwire: FirmWire workspace <Workspace CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4_workspace>
[WARN] firmwire.vendor.shannon.machine: No Ghidra symbol table found. Output will be addresses only
[WARN] firmwire.emulator.firmwire: Memory TOC_BOOT size 0x2e40 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory TOC_BOOT_LOW size 0x2e40 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory TOC_VSS size 0x5e0660 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Peripheral DSPPeripheral at 0x47389c00 is not page aligned. This may causes crashes
[WARN] firmwire.emulator.firmwire: Peripheral DSPPeripheral size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim0 size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim1 start address 0x82008100 is not page aligned. This may causes crashes
[... cut ...]
"quit" from a client, or using Ctrl-\ in UNIX-like environments.

To read more about this, see https://github.com/ipython/ipython/issues/2049

To connect another client to this kernel, use:
    --existing kernel-429.json
[DEBUG] asyncio: Using selector: EpollSelector

`
and then starting jupyer in a separate tab like so
`jupyter console --existing
[ZMQTerminalIPythonApp] CRITICAL | Could not find existing kernel connection file kernel*.json
`

It looks like firmwire should be writing a kernel-429.json config file. But I doesn't seem to be saved anywhere.

Gio-1230, 2023-07-23:
@Miracle-doctor , @chromed-lucy 

hi guys.

Remember, you're working in a docker container.

Check the docker process list.
you can get container id of firmwire.

`host $ docker ps -a
CONTAINER ID   IMAGE                       COMMAND                  CREATED       STATUS       PORTS                                       NAMES
36de91227b5e   firmwire                    "/bin/bash"              3 hours ago   Up 3 hours                                               dreamy_matsumoto
`

2. connect your firmwire container.
`host $ docker exec -it 36de91227b5e /bin/bash`

3. execute jupyter.
root@36de91227b5e # jupyter console --existing