FirmWire/FirmWire #24: MTK movtz instruction incorrect output

jsolmons, 2023-01-26:
This issue should be in the Firmwire Panda repository; however, issues cannot be generated in that repository. Please forgive that I am posting here.

When encountering a MIPSe16 movtz instruction, the output does not appear correct. In the MIPS16e2 spec, there are two different combinations for movtz, one where the rb register is moved and another when rb bits are zero, then the zero register is used. When the zero register is supposed to be used, it appears the s0 register is moved into the destination instead. See code below. Upon additional testing, I also noticed errors when movtz was used with non-zero rb values, for a register to register movtz (e.g. movtz a2, a3; raw bytes 0x27 0xf0 0x16 0x36). I noticed in the source, translate.c "MOVTZ is untested" which is my starting point to trying to fix this issue. If you are able to fix first, great. If not, please advise on how I can solve this issue. Thank you for your time. 

Sample Code:
cmpi v0,0x61  (raw bytes: 0x61 0x72)
li v0,0x8   (raw bytes: 0x08, 0x6a)
movtz v0,zero     (raw bytes: 0x00 0xf0 0x16 0x32)

Expected output
When v0 is 0x61, v0 should be 0x0.
When v0 is not 0x61, 0v should be 0x8.

Current output:
When v0 is 0x61, v0 becomes s0.
When v0 is not 0x61, v0 becomes 0x8

References: 
MIPS16e2 Application-Specific Extension Technical Reference Manual 
https://www.mips.com/products/architectures/ase/ase16e/ (pg 56 and 57)

jsolmons, 2023-01-27:
I think I just fixed this... will be sending a PR in the next 24 hours.

jsolmons, 2023-01-27:
I attempted to perform a PR from the Firmwire/Pandra repo, which directed me to the main panda repo. I cannot do a PR to the firmwire panda repo. Please see attached patch file for that repo.

[mips16e2_movtz.patch.txt](https://github.com/FirmWire/FirmWire/files/10520350/mips16e2_movtz.patch.txt)

mariusmue, 2023-01-28:
Thank you for looking into this and posting the patch! I will look into what is going on with the PANDA repository, and propagate your patch!

mariusmue, 2023-01-28:
Included in firmwire/Panda with https://github.com/FirmWire/panda/commit/88609968c2cd75727e06f2f3322a21bd13199bd2!

Thank you again, now one can also open issues in the panda directory.

closed 2023-01-28 by mariusmue