FirmWire/FirmWire #25: Fuzzing against the Machine

jeppojeps, 2023-01-27:
Hello team, thanks for your awesome work, together with my colleague Eduardo we are about to publish a book which also has some fuzz tests with FirmWire. This follows the lines of the B.Sc of Adrian Hacar Sobrino and Marina Garcia Caro at UC3M Madrid. In principle in the book we bring the user to this point https://twitter.com/adrihacar/status/1412383100580122625 providing a neutral (non-vulnerable) setup message and the code Adrian used for his OTA PoC. 

The book will be published by Packt Publishing and the final TOC is the following:

Chapter 1, Who This Book Is For
Chapter 2, History of Emulation
Chapter 3, Qemu From the Ground
Chapter 4, Qemu Execution Modes and Fuzzing
Chapter 5, A Famous Refrain: AFL+QEMU = CVEs
Chapter 6, Modifying QEMU for basic instrumentation
Chapter 7, Real-life Case Study: Samsung Exynos Baseband, dives into the CVE-2020-25279
Chapter 8, Case Study: OpenWRT full system fuzzing
Chapter 9, Case Study: OpenWRT System Fuzzing for ARM, 
Chapter 10, Finally Here: iOS Full System Fuzzing
Chapter 11, Deus Ex Machina: Fuzzing Android Libraries

please not that we are not sure on the CVE number cause there was one zero-day and one n-day for anomalous SETUP messages, anyways FirmWire harness is explained in detail and we make a walkthrough of the SETUP phase.

I already tagged Marius and Grant on social media.

mariusmue, 2023-01-28:
Hi there!

That is great, is there anything we can do from our side to assist you/do you want us to review the according chapter? (To be frank, I'm also quite interested in the book in general, I was playing with the idea of writing something more about emulation in general, so I'm happy that there will be a book!)

We do have our testcases available, together with the according CVE number in the experiment repository which is still be to be released. Maybe we can discuss this in detail over twitter?

Cheers,
Marius

jeppojeps, 2023-01-28:
Thanks @mariusmue I tagged Grant and asked to start a shared convo so we can talk about on twitter. It seems I can't seems direct messages to you. @jeppojeps is my account.