FirmWire/FirmWire #26: AFL++ gets stuck in forkserver initialization

Mem2019, 2023-02-20:
Hello, we are trying to use FirmWire for fuzzing. Based on the docker image provided, we also compile AFL++ with unicorn mode enabled as shown below.

```
RUN apt-get -y install gdb lz4 cmake
RUN git clone https://github.com/AFLplusplus/AFLplusplus.git /AFLplusplus && \
  cd /AFLplusplus && make clean all && cd unicorn_mode && ./build_unicorn_support.sh
```

However, running command `/AFLplusplus/afl-fuzz -i in/ -o /tmp/out -U -- ./firmwire.py --fuzz gsm_cc --fuzz-input @@ ./modem/modem.bin` according to documentation with large `AFL_FORKSRV_INIT_TMOUT` gets stuck in forkserver initialization for more than 12 hours. I would like to know how such situation can be solved.

Thank you very much.

mariusmue, 2023-02-21:
Hi,

It's likely that your modem is not fully supported and hangs somewhere during bootup/initial emulation. To debug this issue, I would recommend to enable debug output by modifying the commandline as follows:

`AFL_DEBUG_CHILD=1 /AFLplusplus/afl-fuzz -i in/ -o /tmp/out -U -- ./firmwire.py --fuzz-triage gsm_cc --fuzz-input @@ ./modem/modem.bin`

Mem2019, 2023-02-21:
> Hi,
>
> It's likely that your modem is not fully supported and hangs somewhere during bootup/initial emulation. To debug this issue, I would recommend to enable debug output by modifying the [...]

Thanks for replying. The `modem.bin` is decompressed from [here](https://github.com/grant-h/ShannonFirmware/raw/master/modem_files/CP_G973FXXU3ASG8_CP13372649_CL16487963_QB24948473_REV01_user_low_ship.tar.md5.lz4
), which comes from documentation.

In addition, the output  when `AFL_DEBUG_CHILD=1` is shown as follows:

```
*] No auto-generated dictionary tokens to reuse.
[*] Attempting dry run with 'id:000000,time:0,execs:0,orig:input'...
[*] Spinning up the fork server...
              ___            __      _
-.     .-.   | __|(+) _ _ _ _\ \    / /(+) _ _ ___    .-.     .-
  \   /   \  | _|  | | '_| '  \ \/\/ /  | | '_/ -_)  /   \   /
   '-'     '-|_|   | |_| |_|_|_\_/\_/   | |_| \___|-'     '-'
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                A  baseband  analysis  platform
                   https://github.com/FirmWire

[INFO] firmwire.loader: Reading firmware using ShannonLoader (shannon)
[INFO] firmwire.vendor.shannon.loader: SoC <ShannonSOC S5000AP - 20190103> (automatic)
[INFO] firmwire.emulator.patterndb: Searching for patterns in [40010000 - 425579a0]
[INFO] firmwire.emulator.patterndb: Found symbol boot_mpu_table -> 41777400 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_setup_memory -> 40415088 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol boot_key_check -> 40549f8a [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol OS_fatal_error -> 4054ccb2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemAlloc -> 423916b2 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Fixing up TCM region symbol pal_MemAlloc (423916b2 -> 0401581e)
[INFO] firmwire.emulator.patterndb: Found symbol pal_MemFree -> 40cbb1ac [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_MsgSendTo -> 41155e80 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol pal_Sleep -> 40cba1f8 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol log_printf -> 405489ae [CACHED]
[WARN] firmwire.emulator.patterndb: Unable to resolve dynamic symbol log_printf2. Functionality may be affected
[INFO] firmwire.emulator.patterndb: Found symbol pal_SmSetEvent -> 4054d5a2 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_EVENT_GROUP_LIST -> 416f93ec [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Dereference [0x416f93ec] -> 0x418385e8
[INFO] firmwire.emulator.patterndb: Found symbol SYM_TASK_LIST -> 0176f534 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Found likely task name: b'GMAC\x00\x00\x00\x00UL', keeping task layout
[INFO] firmwire.emulator.patterndb: Found symbol SYM_SCHEDULABLE_TASK_LIST -> 43a36e68 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_CUR_TASK_ID -> 418385f4 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_FN_EXCEPTION_SWITCH -> 40c71734 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol SYM_QUEUE_LIST -> 4013d408 [CACHED]
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_SXXXAP_DVFS_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Skipping symbol QUIRK_S337AP_SHM_HACK for S5000AP
[INFO] firmwire.emulator.patterndb: Found symbol SYM_LTERRC_INT_MOB_CMD_HO_FROM_IRAT_MSG_ID -> 0000c3a3 [CACHED]
[INFO] firmwire.emulator.patterndb: Found symbol DSP_SYNC_WORD_0 -> 4060e214 [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Retrieved sync word 0: 141
[INFO] firmwire.emulator.patterndb: Found symbol DSP_SYNC_WORD_1 -> 4060e21a [CACHED]
[INFO] firmwire.vendor.shannon.pattern_handlers: Retrieved sync word 1: 286
[INFO] firmwire.emulator.patterndb: Dynamic symbol resolution took 0.02 seconds
[INFO] firmwire.vendor.shannon.loader: Using blank NV data
[INFO] firmwire.loader: Loading complete
[INFO] firmwire: FirmWire initializing ShannonMachine
[INFO] firmwire.emulator.firmwire: FirmWire workspace <Workspace modem/modem.bin_workspace>
[WARN] firmwire.vendor.shannon.machine: No Ghidra symbol table found. Output will be addresses only
[INFO] firmwire.vendor.shannon.machine: Inputs will be provided via /tmp/out/default/.cur_input
[INFO] firmwire.vendor.shannon.machine: AFL panic address set [0x0,0x4,0xc,0x10,0x4054ccb2]
[WARN] firmwire.emulator.firmwire: Memory TOC_BOOT size 0x2e40 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory TOC_BOOT_LOW size 0x2e40 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory TOC_VSS size 0x5e0660 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Peripheral DSPPeripheral at 0x47389c00 is not page aligned. This may causes crashes
[WARN] firmwire.emulator.firmwire: Peripheral DSPPeripheral size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim0 size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim1 start address 0x82008100 is not page aligned. This may causes crashes
[WARN] firmwire.emulator.firmwire: Memory tim1 size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim2 start address 0x82008200 is not page aligned. This may causes crashes
[WARN] firmwire.emulator.firmwire: Memory tim2 size 0x100 is not page aligned. Force aligning
[WARN] firmwire.emulator.firmwire: Memory tim3 start address 0x82008300 is not page aligned. This may causes crashes
[... cut ...]
[INFO] firmwire: Starting emulator ShannonEMU3334
==> BOOT
AFL_COMPCOV_LEVEL not set.
[INFO] firmwire.hw.peripheral.ShannonSOCPeripheral.SOC: CHIP_ID read: 50000000

s5000ap (Aug  6 2018@==> WAIT SHUTDOWN
13:55:35):SIPCFeG
Mode=424F4F54
#mCMSBoot
[INFO] firmwire.hw.fifo: SHM raw_rx_buff[DEQUEUE] \x00\xd7\x00\x00\x0d\xa0\x00\x00
Ready => [INFO] firmwire.hw.fifo: SHM raw_rx_buff[DEQUEUE] \x00\xd7\x00\x00\x00\xaf\x00\x00
e@Run
[INFO] firmwire.hw.peripheral.ShannonSOCPeripheral.SOC: CHIP_ID read: 50000000
```

Mem2019, 2023-03-06:
The main problem is that breakpoint at `boot_key_check` is not handled. The breakpoint should be handled by callback function set by `avatar.watchmen.add("BreakpointHit", ...)`, but when FirmWire is executed, such callback function is not executed, so the emulation hangs at breakpoint location forever. I am not sure if this is a bug of panda or something else.

mariusmue, 2023-03-08:
Thanks for looking into it more! Please see my comment in #28 for further discussion

bond187, 2023-04-19:
Hello! Was there ever a resolution to this issue? I'm stuck at the same point trying to fuzz a Shannon BP.

mariusmue, 2023-04-19:
The solution here is to create a snapshot as fuzzing base, after initialization.

The reason is that in fuzzing (non-triage) mode, firmwire can not deal with breakpoints, but these are needed during init for some of the basebands. So, the workflow is:
- boot up firmwire with the fuzzing task injected but in triage mode, and the `--snapshot-at` flag to create a snapshot after initialization 
- start the actual fuzzing from the snapshot (using the `--restore-snapshot`) flag

bond187, 2023-04-19:
Great, thank you! How do I know what value to pass to `--snapshot-at`? I'm pretty new to fuzzing so sorry if it's obvious :)

mariusmue, 2023-04-19:
it's a tuple of `snapshot_addr, snapshot_name` - should be part of the CLI documentation. As address, you want to use one after the initialization.

More about snapshots here: https://firmwire.github.io/docs/workspaces.html