FirmWire/FirmWire #32: The format of `modem.bin` of newer baseband firmware image has changed.

Mem2019, 2023-08-30:
The format of newer `modem.bin` of some Samsung devices (e.g.  SM-F711U) seems to be changed to the image of a FAT16 file system. Is it still possible to extract an TOC file from it?

For example:
```
modem.bin: DOS/MBR boot sector, code offset 0x3c+2, OEM-ID "MSDOS5.0", Bytes/sector 4096, sectors/cluster 4, root entries 512, Media descriptor 0xf8, sectors/FAT 7, sectors/track 63, heads 255, sectors 56320 (volumes > 32 MB), serial number 0xbc614e, unlabeled, FAT (16 bit)
```

grant-h, 2023-08-30:
This looks like a Qualcomm Samsung modem, not an Exynos modem (Shannon) which is supported.

Mem2019, 2023-08-30:
> 

Thanks, I will have a look.