FirmWire/FirmWire #33: gdb-remote debugging disconnects on crash

nishant412, 2023-10-19:
Hi,
I am trying to utilize interactive exploration using gdb remote. In particular, I am hoping to replay a crash input using fuzz-triage option and then use gdb-remote to explore the program state at the instant the crash occurs. Unfortunately, gdb-remote disconnects as soon as the crash error is hit (Fatal Error: `PAL_MEM_GUARD_CORRUPTION`). 

Here is what I am trying to do. I start my replay session with the crash input and the `-S` and `-s` flags. When the CPU stops after initialization, I connect through gdb using target remote command, and then continue from gdb hoping for it to catch some signal and pause when the fatal error occurs. Unfortunately, the emulation aborts, and gdb remote is closed.

I might be doing something terribly wrong, would really appreciate any inputs on this.

Thanks in advance.

nishant412, 2023-10-23:
Hi,

Please if someone has any ideas it will be really helpful. In particular I am trying to reproduce the vulnerability CC1 from the paper and run a gdb remote session to look at the memory and registers at the instant the memory error occurs. I am reproducing the error based on the instructions here : https://github.com/FirmWire/ndss22_experiments/tree/main/V-D_vulnerabilities 

Thanks in advance

mariusmue, 2023-10-23:
Hi,

Thank you for your interest in our project and please excuse the delay.

The issue you are encountering is due the fact firmwire is programmed; when an abort is triggered, the emulation is stopped, *especially* for fuzzing modes, as this way AFL knows about the occured crash.

I would recommend you to collect achieved coverage (e.g., via the `--trace-bb-collection` or `--fuzz-triage` commandline argument. In the case of fuzz-triage, you should see the output in the workspace directoy, while trace-bb-collection writes it to stdout). Then, you can set a breakpoint somewhen before the actual crash occurs and single-step.

Alternatively, you could try commenting this line of the shannon-machine: https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/shannon/machine.py#L385

When commented, the panic-addresses are not passed on to panda, and emulation should be continuing, although I didn't test this yet.

nishant412, 2023-10-23:
Thanks a lot @mariusmue that helps a lot.

closed 2023-10-24 by nishant412