FirmWire/FirmWire #34: Issues connecting gef-remote to Firmwire emulation

nishant412, 2023-10-24:
I am trying to replay a crash input using fuzz-triage and then interactively explore using `gef-remote`. I was able to successfully connect plain `gdb remote` to the running emulation and set breakpoints and explore the code, but I wanted a convenient view of the program memory (stack and heap), so was trying to connect using `gef-remote`. It does appear to connect initially but then seems to error out

> gef➤  gef-remote localhost 1234
> warning: No executable has been specified and target does not support
> determining executable automatically.  Try using the "file" command.
> 0x00000000 in ?? ()
> [!] Command 'gef-remote' failed to execute properly, reason: No session started

On the remote side (the emulation started with -s and -S options), I see the following messages

> [INFO] firmwire.vendor.shannon.machine: TASK100: CMMO (0x40f31a49)
> [INFO] firmwire.vendor.shannon.machine: TASK101: AFL_GSM_CC (0x4b000001)
> [INFO] firmwire: Starting emulator ShannonEMU3334
> ==> BOOT
> ==> HALTED
> ==> WAIT SHUTDOWN
> [INFO] avatar2.gdbplugin: Accepted connection from ('127.0.0.1', 33642)
> [CRIT] avatar2.gdbplugin: Received not implemented packet: b'Qqemu.sstepbits'

Really appreciate any inputs on this. I am using the Firmwire docker image. 

Thanks in advance

mariusmue, 2023-10-24:
`gef-remote -q` should be the solution here. (I believe on more recent versions of gef, that is `gef-remote --qemu`)

nishant412, 2023-10-24:
Thanks for the response. Unfortunately this is what I get when I try to add the `--qemu-user` flag

> root@baa8ac80d991:/firmwire# LC_ALL=C.UTF-8 gdb-multiarch -q
> GEF for linux ready, type `gef' to start, `gef config' to configure
> 89 commands loaded and 5 functions added for GDB 9.2 in 0.00ms using Python engine 3.8
> gef➤  gef-remote --qemu-user localhost 1234
> [!] Failed to initialize qemu-user mode, reason: None does not exist
> gef➤  

(I had to add the LC_ALL variable when launching `gdb-multiarch` to solve this `gef` issue : https://github.com/hugsy/gef/issues/195 )

Thanks in advance

nishant412, 2023-10-26:
Hi @mariusmue , any inputs on what might be wrong? Really appreciate the help on this.

Thanks in advance

mariusmue, 2023-10-26:
Hi,

Please understand that I'm quite busy these days. It seems there has been a regression between the firmwire and gef interaction due to updates to gef - I don't have time to get this fixed at the moment.

However, you can use an old version of gef, from quick testing, [2021.04](https://github.com/hugsy/gef/blob/2021.04/gef.py) seems to work.

I can connect using the following commands with above version of gef.
```
gdb-multiarch
gef➤  set architecture arm
gef➤  gef-remote -q 127.0.0.1:1234
```

nishant412, 2023-10-27:
Thanks a lot for your response. I apologize for pestering. The 2021.04 version of gef does work correctly, thank you for the recommendation. I can close the issue now, or if you prefer can leave it open (in case you wanna take a look at the new version of gef later). Let me know whatever is better.

closed 2023-11-27 by nishant412