FirmWire/FirmWire #35: steps for adding new loader

docfate111, 2023-11-24:
hello i have a modem binary that is aarch32 and I can load it in ghidra with your script but I cannot use it with Firmwire - I get an error message about not finding a loader for it. On the wiki it mentions that there is a language for extending the emulator but I don't know how to extend it.

mariusmue, 2023-11-24:
Hi.
I assume this is an exynos/shannon based image, as you reference to our Ghidra scripts?

If this is the case, you most likely will need to add SoC support, for this, you would need to extend the following file:
- https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/shannon/soc/__init__.py

You also potentially need to add or adjust [peripherals](https://github.com/FirmWire/FirmWire/tree/main/firmwire/vendor/shannon/hw), and adjust the [machine](https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/shannon/machine.py) to provide relevant hooks for this SoC.

Furthermore, you may need to add/adjust [patterns](https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/shannon/pattern.py).

Hope that helps, but without information what kind of firmware you looking at, I cannot really say more

docfate111, 2023-11-25:
Ok thank you.
What kind of information? Can I send the firmware?

docfate111, 2024-01-03:
How do I determine  CHIP_ID, SIPC_BASE, SHM_BASE, SOC_BASE , SOC_CLK_BASE,  CLK_PERIPHERAL, and TIMER_BASE? I am trying to emulate [r11](https://developers.google.com/android/images-watch).
[image]
Right now firmwire just says not recognized(I removed the first few bytes so the TOC is the beginning) - if I add a class with a different CHIP ID or something will it try to load? Or should I try and make a class based on the chip that is most similar to it?

mariusmue, 2024-02-18:
Hi @docfate111,

Sorry for the delay! Usually, you would determine the addresses of peripherals via reverse engineering. You will want to add a separate class, yes. I started working on emulating the pixel watch modem myself a while ago, which required also
- additional patterns
-  different task layout
- additional clock peripheral
- Minor changes here and there

Unfortunately, I never fully got support working, which is why I never upstreamed it. If you are still interested in this, I can share my patchset. 

Regarding your specific question, the values in question are (according to my reverserse engineering):
```
    CHIP_ID = 0x03070000
    SIPC_BASE = 0x8F920000
    SHM_BASE = 0x45800000
    SOC_BASE = 0x82000000
    SOC_CLK_BASE = 0x83000000
    TIMER_BASE = SOC_BASE + 0x8000
    ```

Best

Br4v3H3r0, 2024-07-05:
Hi @mariusmue ,
Could you please explain the reverse engineering process which led to finding these addresses above? It would be very helpful for us to extend the support for different firmwares (at least Cortex-R ones)

mariusmue, 2024-07-14:
Hey @Br4v3H3r0,

I would usually follow one of the following ways:
1) Run firmwire with debugging flags enabled, especially the ones revolving on unassigned MMIO addresses
2) Take a firmware supported image and load it in Ghidra, using the Shannonloader and script to automagically rename function. From here, explore the functions which access the different parts/addresses. Then, load the firmware image I'd like to support in Ghidra and run the same scripts. Optionally, I'd use function ID to carry over the names of renamed function. Now, I would search again for the functions accessing these different regions.

Hope that helps!