FirmWire/FirmWire #36: Viewing uart_puts messages in Shannon modkit task

nishant412, 2023-11-27:
Hi, I am trying to understand the lte_rrc fuzzing modkit code. I am trying to replay a crash input in triage mode and observe the Firmwire console messages. I am able to reproduce the crash from the ndss22 repo using the following command

> ./firmwire.py -w ./temp_workspace --fuzz-triage lte_rrc --fuzz-input <path_to_crash_inputs>/LTE_RRC_LRRCConnectionReconfiguration_v890_IEs_PREFETCH_ABORT_G973F_CTD1.bin <path_to_modem_files>/CP_G973FXXS5CTD1_CP15661447_CL18242812_QB30535823_REV01_user_low_ship.tar.md5

In the Firmwire output messages, the only messages corresponding to AFL_LTE_RRC task are the task getting inserted and pal_Sleep and pal_MsgSendTo getting called

> [INFO] firmwire.vendor.shannon.machine: Injecting task <TaskMod 'AFL_LTE_RRC' base 0x4b000000> -> slot 103
> ...
> [INFO] firmwire.vendor.shannon.machine: TASK103: AFL_LTE_RRC (0x4b000001)
> ...
> [42.95542][AFL_LTE_RRC] 0x4b000025 pal_Sleep(200)
> ...
> [45.34014][AFL_LTE_RRC] 0x4b000167 pal_MsgSendTo(LTERRC (15)) - PALMsg<0x0000,  (c3a4) -> PS (0), 16 bytes>

I don't however see any of the uart_puts log messages from the fuzz task code. For example, I can't find this message

> uart_puts("[+] Filling the qitem\n");

Where are these messages logged? Can I redirect these messages to the console output?

Thanks in advance

mariusmue, 2023-12-01:
Hi,

To see these messages, you need to enable debug output log for the fuzzing tasks. Change the following [line](https://github.com/FirmWire/FirmWire/blob/main/modkit/shannon/common.h#L7) to
```C
#define ENABLE_DEBUG_OUTPUT 1
```

Afterwards, recompile, reinject, and you should be able to see the output.

nishant412, 2024-01-18:
Hi @mariusmue , sorry for my delay in testing this out.

I tried changing that `ENABLE_DEBUG_OUTPUT` to 1 in the file you mentioned. I then rebuilt the docker using the command 

> docker build -t firmwire .

Unfortunately, I still don't seem to see the `uart_puts` messages in the Firmwire output. Any suggestions?

Thanks in advance

nishant412, 2024-01-25:
Sorry I realized I needed to rebuild the modkit tasks, not the docker. Am able to see the debug messages now.

Thanks