FirmWire/FirmWire #38: MT6765 issue: [ERROR] firmwire.hw.peripheral.PCCIF_Periph.PCCIF0_MD: PCCIF ring no too large (value: 16, is only: 8)

docfate111, 2024-01-07:
```
AT: 0xdeadbeef  -> 0xdeadbeef   -> 0x00000000
V0: 0x0 -> 0x00000000
V1: 0x0 -> 0x00000000
A0: 0x63df642c  -> 0x63df642c   -> 0x00000000
A1: 0x2986174   -> 0x02986174   -> 0x00000001   -> 0x183c0601   -> 0x00000000
A2: 0x0 -> 0x00000000
A3: 0x69100990  -> 0x69100990   -> 0x00000000
T0: 0x0 -> 0x00000000
T1: 0xb2e8      -> 0x0000b2e8   -> 0x2b0eeb8e   -> 0x00000000
T2: 0x0 -> 0x00000000
T3: 0xda79      -> 0x0000da79   -> 0x409a44f0   -> 0x00000000
T4: 0x90bac5e7  -> 0x90bac5e7   -> 0x786a00ea   -> 0x00000000
T5: 0x90db23d4  -> 0x90db23d4   -> 0x00000034   -> 0x00000000
T6: 0x0 -> 0x00000000
T7: 0x3c386ae   -> 0x03c386ae   -> 0x00000000
S0: 0x0 -> 0x00000000
S1: 0x0 -> 0x00000000
S2: 0x2986174   -> 0x02986174   -> 0x00000001   -> 0x183c0601   -> 0x00000000
S3: 0x1 -> 0x00000001   -> 0x183c0601   -> 0x00000000
S4: 0xd0        -> 0x000000d0   -> 0x00000000
S5: 0x0 -> 0x00000000
S6: 0x2 -> 0x00000002   -> 0x0a183c06   -> 0x00000000
S7: 0x0 -> 0x00000000
T8: 0x8 -> 0x00000008   -> 0x00c00008   -> 0xa076f020   -> 0x00000000
T9: 0x691010a0  -> 0x691010a0   -> 0x00000000
K0: 0xdeadbeef  -> 0xdeadbeef   -> 0x00000000
K1: 0xdeadbeef  -> 0xdeadbeef   -> 0x00000000
GP: 0x9f108000  -> 0x9f108000   -> 0xdededede   -> 0x00000000
SP: 0x63df642c  -> 0x63df642c   -> 0x00000000
FP: 0xc8        -> 0x000000c8   -> 0x00000000
RA: 0x90db2598  -> 0x90db2598   -> 0x64faf030   -> 0x00000000
Stack:
[SP+0x00 == 0x63df642c]: 0xdeadbeef00000000     -> 0xdeadbeef00000000   -> 0x3c060157   -> 0x00000000
[SP+0x08 == 0x63df6434]: 0x00000000     -> 0x00000000
[SP+0x10 == 0x63df643c]: 0x298617402986184      -> 0x298617402986184    -> 0x00000000
[SP+0x18 == 0x63df6444]: 0x6910099000000000     -> 0x6910099000000000   -> 0x3c060157   -> 0x00000000
[SP+0x20 == 0x63df644c]: 0xb2e800000000 -> 0xb2e800000000       -> 0x3c060157   -> 0x00000000
[SP+0x28 == 0x63df6454]: 0xda7900000000 -> 0xda7900000000       -> 0x3c060157   -> 0x00000000
[SP+0x30 == 0x63df645c]: 0x90db23d490bac5e7     -> 0x90db23d490bac5e7   -> 0x786a00ea   -> 0x00000000
[SP+0x38 == 0x63df6464]: 0x3c386ae00000000      -> 0x3c386ae00000000    -> 0x3c060157   -> 0x00000000
[ERROR] firmwire.hw.peripheral.PCCIF_Periph.PCCIF0_MD: PCCIF ring no too large (value: 16, is only: 8)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 522, in run
    handler(message)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/watchmen.py", line 78, in watchtrigger
    ret = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 491, in _handle_remote_memory_write_message
    success = mem_range.forwarded_to.write_memory(
  File "/usr/local/lib/python3.8/dist-packages/avatar2/peripherals/avatar_peripheral.py", line 66, in write_memory
    return intervals.pop().data(offset, size, value, **kwargs)
  File "/firmwire/firmwire/vendor/mtk/hw/PCCIFPeripheral.py", line 492, in hw_write
    assert False
AssertionError
```
Can I simply increase the ring?

mariusmue, 2024-01-07:
Hi,

The ring size is SoC specific; I'm not sure if the SoC you are using has been tested by us.
Either way, for adjusting, I would suggest to change the according SoC definition, located in https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/mtk/soc/__init__.py

The variables to adjust should be `SHM_QUEUE_NUM` and `SHM_LAYOUT`; Our definition for the [MT6765](https://github.com/FirmWire/FirmWire/blob/main/firmwire/vendor/mtk/soc/__init__.py#L87) seems to follow that layout.

If you have a new SoC and it works after adding a definition, we'd happily look forward to a pull request :)

docfate111, 2024-01-07:
How do I determine what goes in SHM_LAYOUT?
I was able to get rid of the error by (appending zeroes and increasing SHM_QUEUE_NUM to 17)[https://github.com/docfate111/FirmWire/commit/bc19ae612bee689bed4791fc72ba22718d4bad22] since the value that is used to index the ringbuffer is 16 but now readPacket() returns None which causes an error.

```
[SP+0x38 == 0x63df6464]: 0x32a836fa00000000     -> 0x32a836fa00000000   -> 0x3c060157   -> 0x00000000
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 522, in run
    handler(message)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/watchmen.py", line 78, in watchtrigger
    ret = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/avatar2/avatar2.py", line 491, in _handle_remote_memory_write_message
    success = mem_range.forwarded_to.write_memory(
  File "/usr/local/lib/python3.8/dist-packages/avatar2/peripherals/avatar_peripheral.py", line 66, in write_memory
    return intervals.pop().data(offset, size, value, **kwargs)
  File "/firmwire/firmwire/vendor/mtk/hw/PCCIFPeripheral.py", line 495, in hw_write
    packet = bytes(ring.readPacket())
TypeError: cannot convert 'NoneType' object to bytes
```

noopwafel, 2024-02-18:
You need to check the corresponding Linux kernel source.

For the MT6765, you can look at https://github.com/xiaomi-mt6765/android_kernel_xiaomi_mt6765/blob/lineage-17.1/drivers/misc/mediatek/eccci/hif/ccci_hif_ccif.c - but the only other values there seem to be equivalent to our MT6768 config. Did you try just copying that one directly? (Or do you have kernel source matching your firmware?)

docfate111, 2024-02-25:
the kernel version is 
```
MT6765 
4.9.190+
QKQ1
Oct 1, 2020 patch level
```

When I try the https://github.com/xiaomi-mt6765/android_kernel_xiaomi_mt6765/blob/lineage-17.1/drivers/misc/mediatek/eccci/hif/ccci_hif_ccif.c#L183C1-L188C2 the error is  `ERROR] firmwire.hw.peripheral.PCCIF_Periph.PCCIF0_MD: PCCIF ring no too large (value: 16, is only: 8)` so I had changed it the 16 instead but then I get
`ERROR] firmwire.hw.peripheral.PCCIF_Periph.PCCIF0_MD: PCCIF ring no too large (value: 16, is only: 16)`