FirmWire/FirmWire #44: Injecting Multiple task

lafreuxpabo, 2024-10-16:
Hello, first of all thanks for your awesome work !
Im currently working on baseband emulation and as I was trying to fuzz some tasks, I tried to inject multiple task in the firmware.
I was surprised to see that it was not natively supported. I modified a little bit the `firmwire.py` file and managed to get 2 tasks injected. I was planning on making a PR but I wanted to ask before, is there any reason that could justify the fact that, natively, firmwire only supports injecting a single task with `-t/--module` option ?
Thanks before all and thanks for you work, have a nice day

mariusmue, 2024-10-16:
Heya,

Thanks for reaching out.

Happy to review a PR on that issue. The main reason why by default only one task can be injected is the logic behind finding a location to inject it. If your solution would allow flexibility to (optionally) choose where to inject each task, I think that official support for multi-task injection should not be an issue.

Best,
Marius

lafreuxpabo, 2024-10-16:
Hey, thanks for your answer, at the moment I simply modified the way `firmwire.py` in function `main` handles the argument `--module`.
Short story: I added a condition to check if a ',' is present in the argument `args.injected_task` to then split it to a list, and add the name of each parsed task to a new list, `asked_modules`. This new list is then processed through a loop to inject each desired task with `machine.load_and_inject_task`

I'll make my PR as soon as possible today
Thanks again and best regards